Mellow-3D/FLYF407ZG #1: Pcb schematics missing

tonn0297, 2019-12-27:
Hello, i wuld like to see the pcb schematics to validate it will fit my use.
can you please upload them in pdf format?

kind regards Tonny

FLYmaker, 2019-12-28:
> 您好，我想看一下PCB原理图以验证它是否适合我的使用。
> 您能以pdf格式上传吗？
> 
> 亲切的问候

[image]
[image]
[image]
[image]

DMCShep, 2019-12-31:
Those are very incomplete, can we get FULL schematics? Also PCB design ("gerber") files.

This is an open source website, but it's got a very "closed source" approach, and your constant price increases aren't helping either.

FLYmaker, 2020-01-01:
> 这些还很不完整，我们可以获得完整的原理图吗？还有PCB设计（“ gerber”）文件。
> 
> 这是一个开放源代码的网站，但是它有一个非常“封闭源”的方法，而且您不断上涨的价格也无济于事。

> 这些还很不完整，我们可以获得完整的原理图吗？还有PCB设计（“ gerber”）文件。
> 
> 这是一个开放源代码的网站，但是它有一个非常“封闭源”的方法，而且您不断上涨的价格也无济于事。

The complete schematic diagram will not be sent out for the time being. The price we sell to dealers has not changed. Their price increase has nothing to do with us.

DMCShep, 2020-01-01:
When can schematics be expected to be released?

FLYmaker, 2020-01-01:
> 何时可以发布原理图？
https://github.com/FLYmaker/FLYF407ZG/blob/master/picture/Pin%20diagram.png
You can see this picture, similar to the schematic diagram

DMCShep, 2020-01-01:
You stated "The complete schematic diagram will not be sent out for the time being." -- I am asking for clarification on this. I have seen what is posted but as I said before it is incomplete. I will rephrase:

When can full, complete schematics be expected to be released?

tonn0297, 2020-01-01:
The licence you have on this states it's open source and so is this platform. So please release the files. A PDF of the full schematic is enugh. Else you are fighting your own licence therms.

DJDOUBLEG, 2020-03-05:
> > 您好，我想看一下PCB原理图以验证它是否适合我的使用。
> > 您能以pdf格式上传吗？
> > 亲切的问候
> 
> [image]
> [image]
> [image]
> [image]

Please upload full schematic diagram!!

Coffee0297, 2020-04-25:
please upload full pcb schematics... this is open source site and atm you are closed source.

DJDOUBLEG, 2020-04-30:
Please please please upload!

pszeder, 2020-08-26:
Please please please upload the schematic!!!

FLYmaker, 2020-08-26:
> 请请上传原理图！！！

https://github.com/FLYmaker/FLYF407ZG/tree/master/picture

pszeder, 2020-08-26:
I've looked through them, but couldn't find it

FLYmaker, 2020-08-26:
> > 请请上传原理图！！！
> 
> https://github.com/FLYmaker/FLYF407ZG/tree/master/picture

> 我看过它们，但找不到

[image]

pszeder, 2020-08-26:
That's not what I'm looking for.  I'm looking for the full document where the photos you inserted above are coming from. 
[image]

FLYmaker, 2020-08-26:
> 那不是我要找的东西。我正在寻找您上面插入的照片来自的完整文档。
> [image]

Isn't this a schematic diagram? What else do you need?  .The pictures above are from the schematic

pszeder, 2020-08-26:
These are just small pieces of it, I would like to have the whole image/PDF. I want to develop a new firmware

FLYmaker, 2020-08-26:
> 这些只是其中的一小部分，我希望获得整个图像/ PDF。我想开发一个新固件

[image]
Please download it and open it

FLYmaker, 2020-08-26:
> 这些只是其中的一小部分，我希望获得整个图像/ PDF。我想开发一个新固件

Sorry I made an upload error, please wait

DJDOUBLEG, 2020-08-26:
We need full schematic diagram, like this: https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/Schematic

FLYmaker, 2020-08-26:
> 我们需要完整的示意图，像这样：[https](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/Schematic) : [//github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/Schematic](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/Schematic)

Sorry to upload the wrong file before, I am uploading again

FLYmaker, 2020-08-26:
[Schematic.pdf](https://github.com/FLYmaker/FLYF407ZG/files/5130680/Schematic.pdf)