Mellow-3D/FLYF407ZG #2: Jumpers under drivers

apsbiker, 2020-01-09:
Any way we can get a hookup guide for the board? There are many jumpers available under the driver board area (far more than the normal 3 jumpers for MS1, MS2, MS3) but no documentation as for what jumper controls what setting. 

[image]

FLYmaker, 2020-01-09:
Default mode

[image]

spi mode
[image]

uart  mode
[image]

naorbonomo, 2020-01-14:
How do you enable sensorless homing? which pins need to be wired where?

Nazrac, 2020-01-15:
I can tell you by the board that sensorless homing will not work. There would be another set of 2 connectors just beside the EN(enable) pin. I am seriously thinking about this but my SKR pro with some ingenuity would be able to support 9 steppers if you get creative with the expansion ports. The worst part of this board is the lack of documentation, along with the lackluster onboard mosfet for the heated bed, but I run mosfets anyways. But realistically you could set this up to run a Ender 5 style cartesian with dual carriages 3 z steppers for auto 3 point leveling and still have room for a dual extruder (chimera/cyclops) on one carriage and a run of the mill V6 or Mosquito on the other. Or a coreXY with a toolchanger setup and 4 extruders still keeping the 3 stepper autoleveling. Thats also assuming you can figure out how to drive the system without all the auxiliary pins.

naorbonomo, 2020-01-15:
I am looking for the DIAG pin todo just that, but i dont know which one of the three it is