Mellow-3D/FLYF407ZG #3: Firmware Build

naorbonomo, 2020-01-14:
Is there any working firmware out there? 

I'm unable to build it in VSCode.

Also tried using the firmware from this branch, no luck!

I'm Getting this error :

collect2.exe: error: ld returned 1 exit status
*** [.pio\build\FLYF407ZG\firmware.elf] Error 1

FLYmaker, 2020-01-14:
Find this line in your firmware and delete it.

https://github.com/FLYmaker/FLYF407ZG/blob/227152684c1872e6a8deb4570b2b5416e9c3c826/%E5%9B%BA%E4%BB%B6/buildroot/share/PlatformIO/boards/FLYF407ZG.json#L17

naorbonomo, 2020-01-20:
> Find this line in your firmware and delete it.
> 
> https://github.com/FLYmaker/FLYF407ZG/blob/227152684c1872e6a8deb4570b2b5416e9c3c826/%E5%9B%BA%E4%BB%B6/buildroot/share/PlatformIO/boards/FLYF407ZG.json#L17

this does not work.

FLYmaker, 2020-01-20:
> > 在固件中找到此行并将其删除。
> > https://github.com/FLYmaker/FLYF407ZG/blob/227152684c1872e6a8deb4570b2b5416e9c3c826/%E5%9B%BA%E4%BB%B6/buildroot/share/PlatformIO/boards/FLYF407ZG.json#L17
> 
> 这行不通。

You can upload your firmware or send it to my email 1058939249@qq.com to help you in real time

DJDOUBLEG, 2020-01-22:
If i use this setting then work well the platformio compiling: 
default_envs = BIGTREE_SKR_PRO
:) :)