Mellow-3D/FLYF407ZG #6: 5V pins only measure 4.6V

naorbonomo, 2020-01-19:
5V pins onboard, all measure 4.6V
Also checked 3.3V they all measure 3.3V

Tried both 12V and 24V power input using a Lab digital DC power supply. 

Tested with nothing else connected, just a board on the table connected to a power supply. 

Bad board? Or am I missing something ?

FLYmaker, 2020-01-20:
I added a diode here for safety
[image]
You can short it. 4.7V becomes 5V

naorbonomo, 2020-01-20:
should i be doing this or 4.6V is acceptable for stuff like PINDA and BL-Touch?

thanks for the quick responses

FLYmaker, 2020-01-20:
> 我应该这样做还是PINDA和BL-Touch之类的东西接受4.6V？
> 
> 感谢您的快速回复

5V devices can be used normally, please rest assured