Mellow-3D/FLYF407ZG #7: FysETC mini 12864 1.2 supplement

DJDOUBLEG, 2020-01-22:
pins_FLYF407ZG.h

#if ENABLED(FYSETC_MINI_12864)
      #define DOGLCD_CS    PE14
      #define DOGLCD_A0    PE12
      //#define LCD_BACKLIGHT_PIN -1
      #define LCD_RESET_PIN PE10   // Must be high or open for LCD to operate normally.
      #if EITHER(FYSETC_MINI_12864_1_2, FYSETC_MINI_12864_2_0)
        #ifndef RGB_LED_R_PIN
          #define RGB_LED_R_PIN PE9
        #endif
        #ifndef RGB_LED_G_PIN
          #define RGB_LED_G_PIN PE8
        #endif
        #ifndef RGB_LED_B_PIN
          #define RGB_LED_B_PIN PE7
        #endif
      #elif ENABLED(FYSETC_MINI_12864_2_1)
        #define NEOPIXEL_PIN    PE9
      #endif
    #endif // !FYSETC_MINI_12864

DJDOUBLEG, 2020-01-22:
[image]

FLYmaker, 2020-01-27:
Thank you for your contribution

DJDOUBLEG, 2020-01-28:
> Thank you for your contribution

You have full schematic? 
Are you planning to upload?

kakou-fr, 2020-02-29:
Did you try to home or move from display ?

I have a BIQU TFT24, it works in touch mode. But in LCD12864 mode, if I try to move or home axis, the board does not respond and the printer just enable the stepper (and don't move it). The other function like temp or fan works ...