Mellow-3D/FLYF407ZG #8: Klipper can not communicate.

DJDOUBLEG, 2020-01-28:
When i upload klipper FW, then can not communicate klipper with MCU.
[image]
Maybe bootloader problem?

FLYmaker, 2020-01-29:
I study how to use

AniInABox, 2020-01-30:
I have the same Issue.
Kevin from Klipper said that this should be analyzed with a Hardware Debugger und seams to be a Bootloader Problem.

kakou-fr, 2020-02-10:
waiting for it too

kakou-fr, 2020-02-13:
Could you plublish the source of the bootloader ?

robustini, 2020-02-14:
Can someone help me?

https://github.com/FLYmaker/FLYF407ZG/issues/9

DJDOUBLEG, 2020-02-15:
> Could you publish the source of the bootloader ?

I need this bootloader too.

AniInABox, 2020-02-16:
Does something happen in the Background, or can i put the Board into trash?

Please give US the Bootloader Code vor the Wiring Diagramm. Wrn want to usw this Board with Klipper and other Firmwares.

DJDOUBLEG, 2020-02-16:
Work with Marlin!

AniInABox <notifications@github.com> ezt írta (időpont: 2020. febr. 16., V
16:28):

DJDOUBLEG, 2020-02-17:
My opinion is: FLYmaker user is chinese. New coronavirus is much more important than FLYboard.

FLYmaker, 2020-02-27:
klipper FW requires DVECT_TAB_OFFSET = 0x8000
  The bootloader writes the klipper FW to the address after 0x8008000

FLYmaker, 2020-02-27:
> 我的意见是：FLYmaker用户是中国人。新的冠状病毒比FLYboard重要得多。

thank you
Thankfully i did not get sick

AniInABox, 2020-02-27:
> 
> 
> klipper FW requires DVECT_TAB_OFFSET = 0x8000
>   The bootloader writes the klipper FW to the address after 0x8008000

What does that mean? I have to configure Klipper to Write to another Start Position?

AniInABox, 2020-02-27:
Here is the Answer from Kevin (Klipper):

Setting a "32KiB bootloader" in "make menuconfig" sets the firmware address to 0x8008000. So, the problem must be something else.

-Kevin

kakou-fr, 2020-03-21:
no new ?

SmoothieAq, 2020-03-26:
@kakou-fr could you please give some details on how you made it work.

I tried:

1. make menuconfig with:   
   - Micro-controller Architecture: STMicroelctronics STM32
   - Processor model: STM32F407
   - Bootloader offset: 32KiB
   - Clock reference: 8 MHz crystal
   - Use USB instead of serial
2. make the klipper.bin
3. format a 8GB sd card on my Windows pc with FAT32
4. Copy the klipper.bin to that
5. Place the sd card in the FLY and connect it with USB to the Raspberry
6. Power off, then power on

When I look in /dev, I can not see anything that looks like an usb port the the FLY
On the FLY I have 3 green leds

I am at loss what to do now, help appreciated :-)

kakou-fr, 2020-03-26:
│ │                           [ ] Enable extra low-level configuration options                                                 │ │
  │ │                               Micro-controller Architecture (STMicroelectronics STM32)  --->                               │ │
  │ │                               Processor model (STM32F407)  --->                                                            │ │
  │ │                               Bootloader offset (32KiB bootloader (SKR-PRO))  --->                                         │ │
  │ │                           [*] Use USB for communication (instead of serial)                                                │ │
  │ │                               USB ids  --->

copy klipper/bin and rename it firmware.bin

SmoothieAq, 2020-03-27:
yeah, forgot that. Now I get an usb-Klipper_xxx in /dev/serial/by-id :-D
Thank you!

I'll try to configure the first parts of klipper this weekend...

SmoothieAq, 2020-03-28:
It is up an running now.
Seems you just have to read the 10 line instruction in generic-flyboard.cfg and do what it says. What with all the negative fuzz??