Mellow-3D/FLYF407ZG #9: How flash?

robustini, 2020-02-14:
It would be useful to know how the firmware is flashed on this board, if you have to use jumpers or something else.
There is no reference to this on your site.

gl40, 2020-03-14:
same problem !

compilation worked fine,
upload failed

```
dfu-util -d vid:pid,0x1EAF:0x0003 -a 0 -s 0x08000000:leave -D "/Users/jean/devel/marlin-mac/Marlin/.pio/build/FLYF407ZG/firmware.bin"
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 1eaf
Match product ID from file: 0003
dfu-util: No DFU capable USB device available
```

rjacobs1969, 2020-03-17:
You should copy the complied firmware.bin on a micro sd card insert it in the cardreader of the board and power on

kakou-fr, 2020-03-22:
compile, copy the firmware.bin on the sd, put the sd on the board, power the board, wait 30s

curtait, 2020-04-08:
Just do it some times. I do it with some other brand boards. Nothing happens. I had wait minutes. Maybe is possible the bootloader does not work. Because plugin the board on pc, northing happen in device management. What is the jumper set, on the board, to do that upload or make the board visible to sistem ? Can you show a video where it works.

FLYmaker, 2020-04-12:
> 只是做几次。我会和其他一些品牌委员会一起做。什么都没发生。我等了几分钟。引导加载程序可能无法正常工作。由于在PC上插入了电路板，因此在设备管理中会发生问题。板上设置的跳线是什么，以进行上载或使板对系统可见？您可以在视频播放的地方播放它吗？

Put the firmware.bin into the memory card, insert the motherboard, restart the motherboard, wait for 30s, the usb port will usually change, and the firmware.bin will become fly.cur, if the port has not changed, please install the relevant driver

curtait, 2020-04-15:
NO jumper set on the board ? No light expect blink, in that 30s ? what is the name of the driver ? Pls let me know.

curtait, 2020-04-16:
JUST DO IT.
No jumper on the board. No expect led blink.
Connect a display : that wil be your visive signal 
Make sure the SD card is little (up to 4 Gb) in FAT and active.
I had put the file in the SD : "firmware.bin"
Once the board load the firmware give back file : "FLY.cur"

curtait, 2020-04-16:
I had use 
en.stsw-stm32102.zip
Driver

vmax1145, 2021-03-25:
Very inconvinient to copy firmware on SD and flash from SD each time you need to reconfigure marlin
Flash directly from platformIO is MUST HAVE