Mellow-3D/FLYF407ZG #10: You reaaly need to post more details about your board (if you want to sold it) !!!

kakou-fr, 2020-02-23:
I just mount this board with a mixing extruder, 2 Z stepper, switch endstop for X/Y and TOUCH'MI level bed sensor, but it was really hard to find how :
- to connect correctly the endstop (in the pinout, you give the endstop signal, but they are 3 pins, which one is one ????)
- how to conect correctly the fan, in the pinout, the +/- is not provided
- how to connect a servo, 3 pins, no information int the pinout
- for standalone drivers 4 jumper ms1 ms2 ms3, but the 4th jumper is for what ???
- what is the purpose of bt0 bt1 .. pin ????

I brought one of this board, but i don't will buy other as long as you don't give:
- a good pinout or the pcb schematics
- the bootloader sources (because i must use marlin as klipper does not work)

FLYmaker, 2020-02-25:
Sorry, I ca n’t visit here a few days ago, I will follow the perfect information you said in the near future, thank you very much

FLYmaker, 2020-02-25:
Network problem a few days ago

FLYmaker, 2020-02-28:
https://github.com/FLYmaker/FLYF407ZG/blob/master/picture/Illustrating.png

kakou-fr, 2020-03-02:
Some comments : 
1) I don't understand how to connect the "optimal limit switch voltage"
2) I don't understand the purpose of BT0 BT1 (I already asked, but the response don't help me)
3) servo pin number is missing
4) what is UD+ UD-
5) SDA SCL is on the pin on a chipset ?

rjacobs1969, 2020-03-11:
My two cents

1. Most endstops are designed for 5V, so a jumper on the left two pins

3. SERVO0_PIN   =  PE11 (according to https://github.com/FLYmaker/FLYF407ZG/blob/master/English-firmware/Marlin/src/pins/stm32/pins_FLYF407ZG.h )

5. That seems to be the case

bergyla, 2020-04-19:
There are more data missing:
Outputs: max Amp´s respective RDSon (I fried mine BED FET within just a view seconds ...)

Mounting hole dimensions: I could see in a picture that they are X spaced 144.05mm in but what about Y spacing and Diameter...

kakou-fr, 2020-04-19:
> My two cents
> 
> 1. Most endstops are designed for 5V, so a jumper on the left two pins
>    2 BT0 is for entering in DFU mode, BT1 probably something simular
> 2. SERVO0_PIN   =  PE11 (according to https://github.com/FLYmaker/FLYF407ZG/blob/master/English-firmware/Marlin/src/pins/stm32/pins_FLYF407ZG.h )
>    4 UD- UD+ = USB pins for bootloader programing
> 3. That seems to be the case

1. yes i know, i use optical endstop. but they is 3 pins on the jumper, so need to indicate how to use them correctly

2. in the diagram it's better for reading

Pandoriaantje, 2020-05-20:
> There are more data missing:
> Outputs: max Amp´s respective RDSon (I fried mine BED FET within just a view seconds ...)
> 
> Mounting hole dimensions: I could see in a picture that they are X spaced 144.05mm in but what about Y spacing and Diameter...

All holes seem to be same offset from the board edge, so you need to deduct 5.98 form total Y length (the same difference in X). So that would be 105.3mm for center to center in Y.