Mellow-3D/FLYF407ZG #11: bootloader problem ...

kakou-fr, 2020-03-02:
Hi,

1) your bootloader does not work with klipper !!!
2) I experience some problem with Marlin : 
- sometime reset the board does not reset it, I need to unplug completely the power wait 5s and replug to reset/reboot the board
- sometime when plug power, the board doesn't boot, I need to unplug/replug power 1, 2 ... 5, ... 10 times for the board to boot correctly

FLYmaker, 2020-03-14:
Gùjiàn shàngchuán yǐhòu, qǐng jiāng fly.Cur shānchú
18/5000
After the firmware is uploaded, please delete fly.cur

FLYmaker, 2020-03-14:
After the firmware is uploaded, please delete fly.cur

rjacobs1969, 2020-03-14:
is there any manual on how to flash this board?
Currently it's an expensive paperweight as I have no idea how to flash it

housewright30, 2020-03-22:
> is there any manual on how to flash this board?
> Currently it's an expensive paperweight as I have no idea how to flash it

I am having the same issue. I can't find any instruction on how to flash the board.

FLYmaker, 2020-03-27:
> 是否有任何有关如何刷新此板的手册？
> 目前这是一个昂贵的镇纸，因为我不知道如何刷新它

After compiling marlin2.0, a firmaware.bin will be generated. After inserting the file into your memory card and inserting it on the motherboard, power the motherboard. After a few seconds, the firmware will be automatically uploaded to the chip.

FLYmaker, 2020-03-27:
> 是否有任何有关如何刷新此板的手册？
> 目前这是一个昂贵的镇纸，因为我不知道如何刷新它

After compiling marlin2.0, a firmaware.bin will be generated. After inserting the file into your memory card and inserting it on the motherboard, power the motherboard. After a few seconds, the firmware will be automatically uploaded to the chip.

> > 是否有任何有关如何刷新此板的手册？
> > 目前这是一个昂贵的镇纸，因为我不知道如何刷新它
> 
> 我有相同的问题。我找不到任何有关如何刷新电路板的说明。

After compiling marlin2.0, a firmaware.bin will be generated. After inserting the file into your memory card and inserting it on the motherboard, power the motherboard. After a few seconds, the firmware will be automatically uploaded to the chip.