Mellow-3D/FLYF407ZG #12: compile problem with tmc

hexxter, 2020-04-11:
Marlin/src/module/stepper/trinamic.h:29:10: fatal error: TMCStepper.h: No such file or directory

in the ENV is the lib disabled. How can ich compile it with tmc driver support?

BR Hexxter

FLYmaker, 2020-04-12:
#
# FLYF407ZG
#
[env:FLYF407ZG]
platform          = ststm32
board             = FLYF407ZG
platform_packages = framework-arduinoststm32@>=3.107,<4
build_flags       = ${common.build_flags}
  -DSTM32F4 -DUSBCON -DUSBD_USE_CDC -DUSBD_VID=0x0483 -DUSB_PRODUCT=\"STM32F407ZG\"
  -DTARGET_STM32F4 -DVECT_TAB_OFFSET=0x8000
  -IMarlin/src/HAL/STM32
build_unflags     = -std=gnu++11
extra_scripts     = pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
lib_ignore        =SlowSoftI2CMaster, SoftwareSerial
src_filter        = ${common.default_src_filter} +<src/HAL/STM32>

platformio.ini
You can modify the fly part configuration in platformio.ini so that it can be used

gsuresh2u, 2020-04-22:
Same problem for me too, TMCStepprer.h is available in my libaries
`* CLI  > platformio lib search "header:TMCStepper.h"
* Web  > https://platformio.org/lib/search?query=header:TMCStepper.h`

gsuresh2u, 2020-04-22:
@FLYmaker , the above code is already present in platformio.ini
but still its not taking TMC drivers