Mellow-3D/FLYF407ZG #16: SDCard Display versus Onboard - TMC2130

bergyla, 2020-04-14:
I focused several issues and actually trying to figure them out.
My system:

All steppers drivers are TMC2130 (SPI)
SDCARD Support

Now I´m focusing the following Issues:

When setting #define SDCARD_CONNECTION ONBOARD then I will get Error on the LCD: Connection Errors on the TMC_Steppers.
When setting to LCD, then there are no TMC Connection Errors anymore, all steppers are running fine. Fortunately the Onboard SDCard is still used.
But The CD-Change of the LCD-Display is monitored...

SDCard on the LCD is not reachable at any time..

rjacobs1969, 2020-04-17:
Strange, I also use (6x)TMC2130 in SPI mode and use the SD Card reader of the display
Note that I did slowed down the SD card SPI speed (I don't know if it also affect the SPI speed to the 2130's) 

#define SDSUPPORT
#define SDCARD_CONNECTION LCD
/**
 * SD CARD: SPI SPEED
 *
 * Enable one of the following items for a slower SPI transfer speed.
 * This may be required to resolve "volume init" errors.
 */
#define SPI_SPEED SPI_HALF_SPEED

bergyla, 2020-04-17:
> Strange, I also use (6x)TMC2130 in SPI mode and use the SD Card reader of the display
> Note that I did slowed down the SD card SPI speed (I don't know if it also affect the SPI speed to the 2130's)

Kind of strange.
I´m wondering that even with define #define SDCARD_CONNECTION LCD I could see the content of the onboard SD_Card, so I do not asume to have some speed issues of the SPI-Bus.

However I may have Speed issues, while I also find some issues while using onboard SPI with SD_Card:

Sometimes the head is moving to X-/X+ max and I also saw Y- and thenafter continueing as normal.
Also simetimes the print stopped for some seconds, and most strange, I saw that some strings were forgotten at all, but others placed twice...

That were the point I stopped all testing with SD-Card and bought a intelligent Touchdisplay...

gsuresh2u, 2020-04-22:
Hi, i am also using TMC2130, but when compiling, its giving error
`Marlin\src\module/stepper/trinamic.h:29:10: fatal error: TMCStepper.h: No such file or directory`
but TMCStepper library is available 
what may be the problem ?

bergyla, 2020-04-22:
For some (outdated) reason, the TMCStepper Library is added to the ignore List in the FLYF407ZG section of the platformio.ini file.

Please search for the entry: [env:FLYF407ZG]
Some lines beyond you will find the lib_ignore line.
Remove the TMCStepper, from this and the code should compile just fine.

As found out there were some incompatibilities between STM32 and tmcstepper library in 2019 but where fixed several months ago without reactivate tmcstepper again...

gsuresh2u, 2020-04-22:
@bergyla , Thanks removing TMCStepper from [env:FLYF407ZG] in file platformio.ini worked fine
now its compiling well

gsuresh2u, 2020-04-22:
Hi, is this board supports Optical end stop?
If so, how to connect 3 end stops to single port on board ? in parallel ?

bergyla, 2020-04-22:
The board supports up to 6 endstops (Min, max) for each Axis.
Furthermore all endstop connectors are 3 Pin connectors: Signal, GND and a Pin which could be switched between 5V and VCC (12V or 24V) with a seperate jumper common for all endstop connectors.
But caution! As far I could see there is no protection on the Signalpins, so shortcutting to >5V will destroy the Microcontroller.

You could use optical endstops seperatly.
Eventually you must protect your sensors from ambient light...

gsuresh2u, 2020-04-23:
Thanks, mistakenly i asked for end stops, but i meant to ask for filament sensor port
this board has 1 filament sensor port only
in my build , i want to use 3 extruders with 3 in 1 out hot end
so need to connect 3 filament sensors, can i connect them in parallel ?
so that if any filament is out  (or break), it will pause printing
and hot to add power loss detection to this board ?
my build size is 700 * 900, using 3 steppers for Z axis, 2 steppers for Y axis, 1 for X axis
and 3 steppers for 3 extruders

bergyla, 2020-04-23:
Hm, your setup seems to be very similar to mine (2*Z-Axis and also 3 Extruders).
In general, yes you could connect 3 sensors in parallel and connect them to the single filament sensor port.
With mechanical switches NO closing to GND, this should be fine.
If you are thinking to use Active Sensors you have to be aware that there is no disrete Pull-Up resistor but uC internal PullUp will be used by default. This Pull-Up is limited in current and so eventually using external Pullup is the better idea in reliability and hardware(uC) protection...

gsuresh2u, 2020-04-23:
Ok, Thanks for info, what is the POWER_LOSS_PIN to use Power loss recovery ?
i cant see PIN no in any images
and if you want to use more steppers, you can follow this image
[image]

bergyla, 2020-04-23:
Wiring parallel stepper onto one Driver will end up in bad Microstep Performance (or could fail in total depend on your mechanics).
So I would assume to use one Driver per Stepper will be much more satisfying.

But: You are sure that your Z-Axis might be calibrated well all the time?
Marlin supports to control up to 4 Z-Axis Motors and calibrate them with help of an aditional (moving) sensor connected to your printing-Head...