Mellow-3D/FLYF407ZG #20: Compile Error while using Z4

gsuresh2u, 2020-07-10:
Hi, i want to use 4 steppers for Z axis
here is my setup :
Z Axis : 4 steppers
Y Axis : 2 steppers in parallel (no empty port)
X axis : 1 stepper
E0, E1, E2 : 3 steppers
total 9 steppers, are are TMC5160 drivers
using Inductive probe for Z min end stop and for auto leveling
when i try to use 4 steppers for Z axis, it giving compile error
what i am doing wrong ? please suggest
[image]

FLYmaker, 2020-07-10:
Which firmware version？

gsuresh2u, 2020-07-10:
which is same downloaded from this git
#define CONFIGURATION_H_VERSION 020004

FLYmaker, 2020-07-10:
Please use marlin2.0.5.3 version

gsuresh2u, 2020-07-10:
Board name is same ?
```
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_FLYF407ZG  
#endif
```
when i try above, its saying :"Marlin\src\HAL\AVR\../../inc/../pins/stm32f4/pins_FLYF407ZG.h:25:4: error: #error "Oops! Select an STM32F4 board in 'Tools > Board."

FLYmaker, 2020-07-10:
platformio.ini

[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = FLYF407ZG

FLYmaker, 2020-07-10:
* :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT -1

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 1

/**
 * This setting determines the communication speed of the printer.
 *
 * 250000 works in most cases, but you might try a lower speed if
 * you commonly experience drop-outs during host printing.
 * You may try up to 1000000 to speed up SD file transfer.
 *
 * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
 */
#define BAUDRATE 250000

// Enable the Bluetooth serial interface on AT90USB devices
//#define BLUETOOTH

// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_FLYF407ZG
#endif

FLYmaker, 2020-07-10:
https://github.com/FLYmaker/FLYF407ZG/blob/c00177e4a24d1b003572dc2d17a2482cab386528/English-firmware/platformio.ini#L21

gsuresh2u, 2020-07-10:
Thanks , its compiling fine
Need some clarification of connecting steppers
X - X stepper
Y - Y steppers (2 in parallel)
Z - Z1 stepper
E0 - E0 stepper
E1 - E1 stepper
E2 - E2 stepper
E3 - Z2 stepper ?
E4 - Z3 stepper ?
E5 - Z4 stepper ?
Total 9 ports, 10 motors (2 in parallel)
is above connections are right ?

FLYmaker, 2020-07-10:
[image]
You can open the motherboard pin definition file to modify

FLYmaker, 2020-07-10:
If E1 is used as Z2, change all E1 motor pins to Z2》 as shown
[image]

gsuresh2u, 2020-07-10:
Ok, Thanks

gsuresh2u, 2020-07-24:
Hi, please check the pins config
i want to use pins for
X for X
Y for Y
Z for Z1
E0 for Z2
E1 for Z3
E2 for Z4
E3 for E0  (Extruder 1)
E4 for E1 (Extruder 2)
E5 for E2  (Extruder 3)

`#define X_STEP_PIN                          PB9
#define X_DIR_PIN                           PE0
#define X_ENABLE_PIN                        PE1
#ifndef X_CS_PIN
  #define X_CS_PIN                          PG13
#endif

#define Y_STEP_PIN                          PB8
#define Y_DIR_PIN                           PG11
#define Y_ENABLE_PIN                        PG12
#ifndef Y_CS_PIN
  #define Y_CS_PIN                          PG10
#endif

#define Z_STEP_PIN                          PA8
#define Z_DIR_PIN                           PD6
#define Z_ENABLE_PIN                        PD7
#ifndef Z_CS_PIN
  #define Z_CS_PIN                          PD5
#endif

//Z2
#define E0_STEP_PIN                         PD14  //PC7
#define E0_DIR_PIN                          PG4   //PD3
#define E0_ENABLE_PIN                       PG5   //PD4
#ifndef E0_CS_PIN
  #define E0_CS_PIN                         PG3   //PD1
#endif

//Z3
#define E1_STEP_PIN                         PD13    //PC6
#define E1_DIR_PIN                          PD11    //PA15
#define E1_ENABLE_PIN                       PG2     //PD0
#ifndef E1_CS_PIN
  #define E1_CS_PIN                         PD10    //PA14
#endif

//Z4
#define E2_STEP_PIN                         PD12    //PD15
#define E2_DIR_PIN                          PD8     //PG7
#define E2_ENABLE_PIN                       PD9     //PG8
#ifndef E2_CS_PIN
  #define E2_CS_PIN                         PB12     //PG6
#endif

//E0
#define E3_STEP_PIN                         PC7  // PD14
#define E3_DIR_PIN                          PD3   //PG4
#define E3_ENABLE_PIN                       PD4   //PG5
#ifndef E3_CS_PIN
  #define E3_CS_PIN                         PD1   //PG3
#endif

//E1
#define E4_STEP_PIN                         PC6   //PD13
#define E4_DIR_PIN                          PA15  //PD11
#define E4_ENABLE_PIN                       PD0   //PG2
#ifndef E4_CS_PIN
  #define E4_CS_PIN                         PA14    //PD10
#endif

//E2
#define E5_STEP_PIN                         PD15    //PD12
#define E5_DIR_PIN                          PG7     //PD8
#define E5_ENABLE_PIN                       PG8     //PD9
#ifndef E5_CS_PIN
  #define E5_CS_PIN                         PG6     //PB12
#endif
`

FLYmaker, 2020-07-26:
#define X_STEP_PIN                          PB9
#define X_DIR_PIN                           PE0
#define X_ENABLE_PIN                        PE1
#ifndef X_CS_PIN
  #define X_CS_PIN                          PG13
#endif

#define Y_STEP_PIN                          PB8
#define Y_DIR_PIN                           PG11
#define Y_ENABLE_PIN                        PG12
#ifndef Y_CS_PIN
  #define Y_CS_PIN                          PG10
#endif

#define Z_STEP_PIN                          PA8
#define Z_DIR_PIN                           PD6
#define Z_ENABLE_PIN                        PD7
#ifndef Z_CS_PIN
  #define Z_CS_PIN                          PD5
#endif

#define Z2_STEP_PIN                         PC7
#define Z2_DIR_PIN                          PD3
#define Z2_ENABLE_PIN                       PD4
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN                         PD1
#endif

#define Z3_STEP_PIN                         PC6
#define Z3_DIR_PIN                          PA15
#define Z3_ENABLE_PIN                       PD0
#ifndef Z3_CS_PIN
  #define Z3_CS_PIN                         PA14
#endif

#define Z4_STEP_PIN                         PD15
#define Z4_DIR_PIN                          PG7
#define Z4_ENABLE_PIN                       PG8
#ifndef Z4_CS_PIN
  #define Z4_CS_PIN                         PG6
#endif

#define E0_STEP_PIN                         PD14
#define E0_DIR_PIN                          PG4
#define E0_ENABLE_PIN                       PG5
#ifndef E0_CS_PIN
  #define E0_CS_PIN                         PG3
#endif

#define E1_STEP_PIN                         PD13
#define E1_DIR_PIN                          PD11
#define E1_ENABLE_PIN                       PG2
#ifndef E1_CS_PIN
  #define E1_CS_PIN                         PD10
#endif

#define E2_STEP_PIN                         PD12
#define E2_DIR_PIN                          PD8
#define E2_ENABLE_PIN                       PD9
#ifndef E2_CS_PIN
  #define E2_CS_PIN                         PB12
#endif

closed 2020-08-27 by gsuresh2u