Mellow-3D/FLYF407ZG #23: Connection Problem with Pronterface

gsuresh2u, 2020-07-26:
Hi, i tried to connect with Pronterface
Its showing connected, but giving error 
`Starting USB host...Revision register appears incorrect on MAX3421e initialization. Got 255 failed`
i already installed USB drivers, in Devise manager, its showing fine
Please check the attached images
[image]

enabled Software SPI and changed pins and added Jumpers for SPI mode
[image]

and is my Drivers Orientation is Correct same as in the below image ?
[image]

I have connected only Board, not connected anything for testing
(as my printer is Big , its unable to move to indoor)
[image]

FLYmaker, 2020-07-26:
Qǐng bǎ nǐ de gùjiàn fā shànglái, bìngqiě tmc2209 shì bùnéng shǐyòng spi de
28/5000
Please send your firmware up, and tmc2209 cannot use spi

FLYmaker, 2020-07-26:
Please send your firmware up, and tmc2209 cannot use spi

gsuresh2u, 2020-07-26:
Sir, here i attached my Configurations and pins

[pins_FLYF407ZG.zip](https://github.com/FLYmaker/FLYF407ZG/files/4977702/pins_FLYF407ZG.zip)

[platformio.zip](https://github.com/FLYmaker/FLYF407ZG/files/4977703/platformio.zip)
[Configurations.zip](https://github.com/FLYmaker/FLYF407ZG/files/4977704/Configurations.zip)

gsuresh2u, 2020-07-26:
[firmware.zip](https://github.com/FLYmaker/FLYF407ZG/files/4977744/firmware.zip)

FLYmaker, 2020-07-26:
Do you want to use a USB flash drive? That's why this feature is turned on?

gsuresh2u, 2020-07-26:
Yes, want to usb flash drive

FLYmaker, 2020-07-26:
#define USE_UHS3_USB        
Open this to try

FLYmaker, 2020-07-26:
The USB port of the motherboard can also use a USB flash drive, but I haven't tested it yet. I will test it shortly

gsuresh2u, 2020-07-26:
When i enable #define USE_UHS3_USB, its giving compile errors, 
and tried with disabling #define USB_FLASH_DRIVE_SUPPORT 
and uploaded firmware
Now, its connected fine

gsuresh2u, 2020-07-26:
can i need to disable software SPI in marlin ?
i have changed Jumpers for TMC2209 to UART mode

FLYmaker, 2020-07-26:
Yes, you need to close it and use uart

gsuresh2u, 2020-07-26:
Ok, Thanks for support

closed 2020-07-26 by gsuresh2u