Mellow-3D/FLYF407ZG #24: Any real case use on 3d printer?

RemyBaligant, 2020-07-31:
Hi,

Are you sure about the pinout? I can't get direction to work on X driver and other suspect stuff elsewhere(fans and thermistor pinout diffferent on board and marlin).
Why don't you release full schematics like other board?
It would be more easy for users.
Thanks. Best regards

[image]

FLYmaker, 2020-07-31:
[image]
[image]
[image]
The fan pin has been repaired in my github, and today I have submitted pr to marlin

FLYmaker, 2020-07-31:
[image]
Here is the direction of the standard plug drive
I will send the schematic to my github within 2 weeks

RemyBaligant, 2020-07-31:
Thanks for your reply. Really good news for schematics. That was the only thing missing to make this card a best seller.
About the pinout, i mean PE0 is dir pin for X driver. But when i test my cinematics it seems that the X driver is not receiving DIR change. I tried with other tmc2209 same problem. All the rest is working so far...can't test further without being able to g28 successfully.

FLYmaker, 2020-07-31:
Y Z  E0 Has it been tested?

RemyBaligant, 2020-07-31:
Yes these driver are working

FLYmaker, 2020-07-31:
[image]
Check whether the voltage of this transistor is normal (compared to other drivers)
[image]
[image]

Check if these two points are connected

RemyBaligant, 2020-07-31:
E4 not working. No step, no motor movement too.
I tested with ohmmeter and no continuity between the two point. I had already tried to check with stm datasheet.
So my board is defect?
Can you tell me wich pin on stm for step E4? My guess is no continuity too

FLYmaker, 2020-07-31:
[image]
[image]
[image]

FLYmaker, 2020-07-31:
Your motherboard may have problems with soldering, you can find a dealer to exchange for a motherboard. If you have the ability, you can re-solder these pins with electric Luo iron

RemyBaligant, 2020-07-31:
Yes bad soldering, i try to exchange to board. Thanks a lot for your support.
Maybe later i will have some more questions :)

RemyBaligant, 2020-07-31:
For servo connector. Left to right i have : stm out, vcc, gnd.
Right ? 
But when i send M280 P0 S10 for example, bltouch pin won't move.
I used SERVO0_PIN PE11. Is it good pin for pwm out ?

RemyBaligant, 2020-07-31:
[image]

FLYmaker, 2020-07-31:
Please use the English firmware of my github to fix the servo

RemyBaligant, 2020-07-31:
Ok i give a try

RemyBaligant, 2020-07-31:
[image]

I downloaded your latest marlin release.
Wich one is good ? the pinout image or pins setting of marlin ?

FLYmaker, 2020-07-31:
https://github.com/FLYmaker/FLYF407ZG/tree/master/English-firmware

RemyBaligant, 2020-07-31:
I already downloaded it and it seems there is errors. Check by yourself: 
It's not the picture pinout
https://github.com/FLYmaker/FLYF407ZG/blob/master/English-firmware/Marlin/src/pins/stm32f4/pins_FLYF407ZG.h

FLYmaker, 2020-07-31:
Sorry to update again, wait a moment

RemyBaligant, 2020-07-31:
No problem :)

FLYmaker, 2020-07-31:
OK, please download again

RemyBaligant, 2020-09-07:
I received a new card from Mellow, everything running fine now. The first one had bad solderings...
And thanks for the schematics 👍 
Issue can be closed