Mellow-3D/FLYF407ZG #27: How to connect TFT70 V3.0?

gsuresh2u, 2020-08-27:
Today I received TFT70 V3.0
It has 5V input pins to power up screen 
But I have not found any 5V output pins in this board 
Found 3 power output ports in bottom right side
But in description it's saying they are 12V/24V
My input is 12V with 400W PSU
From where I can take 5V output?
Here I am attaching photos 
[image]
[image]

FLYmaker, 2020-08-27:
[image]

FLYmaker, 2020-08-27:
[video.zip](https://github.com/FLYmaker/FLYF407ZG/files/5134705/video.zip)

FLYmaker, 2020-08-27:
This is our own 7-inch screen, you can take a look

gsuresh2u, 2020-08-27:
OK, thanks

closed 2020-08-27 by gsuresh2u