Mellow-3D/FLYF407ZG #28: Polarity Confusion on Endstop pins

gsuresh2u, 2020-09-01:
Sir, today i started assembling my printer'
first connected Filament sensor to port DLJC, but led in that filament sensor is not glowing
then i checked voltage of pins with multimeter , getting 3.3V
but not the pins as shown in documentation
am i missing anything ?
here is the below pins i am getting 3.3V
and what is Optional limit switch Voltage is used for ?do i need to add any jumper on it ?
i want to use PNP capasitive proximity sensor (10 mm distance) as all axis endstops
Plz check the below image
[image]

FLYmaker, 2020-09-01:
[image]

FLYmaker, 2020-09-01:
[image]

FLYmaker, 2020-09-01:
[image]

FLYmaker, 2020-09-01:
You don’t need to close it after the problem is solved, you can leave it to others for reference

gsuresh2u, 2020-09-01:
Ok, Thanks, i have not added any jumper , so i am not getting nay voltage on endstop pins
i want to use 10mm capacitive proximity sensor, it need 6-36V, my input is 12V
do i need to add 7805 mosfet to decrease voltage going to Signal pin ?
[image]

like the below image ?
[image]

gsuresh2u, 2020-09-02:
Added a jumper to select 5V, but i am not getting any voltage on endstop pins
and checked Continuity between middle pin of voltage selector and + pin of endstops, there is no Continuity
am i missing anything ? or board is defective ?
plz check the below images
[image]
[image]

FLYmaker, 2020-09-03:
If the 5vled light is on, it is impossible, you need to check

gsuresh2u, 2020-09-03:
There is no light On

FLYmaker, 2020-09-03:
No way, you first unplug all the drivers, then power on and test

FLYmaker, 2020-09-03:
Didn’t you find your dealer for a new motherboard? This kind of thing shouldn’t happen. Did you connect a high voltage to the signal pin during the test, which would damage the chip。

FLYmaker, 2020-09-03:
Didn't you find your dealer to replace the motherboard with a new one. This kind of thing shouldn't happen, is it because you connected a high voltage to the signal pin during the test, which would damage the chip.

FLYmaker, 2020-09-03:
Didn't you find your dealer to replace the motherboard with a new one. This kind of thing shouldn't happen, is it because if you connected a high voltage to the signal pin during the test, which would damage the chip.

gsuresh2u, 2020-09-03:
I have not connected any Higher voltage , just asked in above posts, i should add 7805 voltage regulator
to use 12V proximity sensor , right ?
and my X and Y axis also not working, asked dealer to replace board, not got any reply
here is the video i am testing
http://www.mediafire.com/file/452h5oq6vsiyn0y/FLY+Controller.mp4/file

FLYmaker, 2020-09-03:
I contacted your dealer, I have re-sent you a piece before, it should arrive this week or next week

FLYmaker, 2020-09-03:
For other sensors, you need to understand its principle and how to use it, please use high voltage sensors with caution

gsuresh2u, 2020-09-03:
Ok, thank you so much

gsuresh2u, 2020-09-03:
If i use any High voltage sensors, i should use 7805 mosfets like this in below link, am i right ?
https://reprap.org/wiki/Z_probe

FLYmaker, 2020-09-03:
Which sensor do you use and is there any principle of use

gsuresh2u, 2020-09-03:
Here is the sensor link
want to use them for X, Y, Z min endstops
https://www.aliexpress.com/item/1081937429.html?spm=a2g0s.9042311.0.0.27424c4dpb9sOt

FLYmaker, 2020-09-03:
Hello, please send your firmware source code here. I will check if your xy is a hardware problem or a software problem. My dealer told me that your address is India, and the reissued address is not yours. So your motherboard has not been reissued yet. And currently cannot be shipped from China to your country

FLYmaker, 2020-09-03:
I checked with the dealer again, and I can send a new motherboard to you. I am communicating with the dealer for other issues.

gsuresh2u, 2020-09-03:
Plz check the attached Configurations
If its software problem, i am happy, no need to wait for long time to receive board
and i purchased 2 boards yesterday
[Configurations.zip](https://github.com/FLYmaker/FLYF407ZG/files/5167074/Configurations.zip)

gsuresh2u, 2020-09-29:
Sir, Today i have received the replacement board along with 1 extra board for backup purpose
as told earlier, one problem is solved, getting continuity between middle pin of voltage selector and positive pin pf Endstop pins
But my other problem remains same, X and Y axis ports are still not working
i have tried with switching driver boards with working ports, problem remains same
i think i am missing something in firmware
i am using ports like below
X = X , Y = Y , Z = Z1 , E0 = Z2 , E1 = Z3 , E2 = Z4 , E3 = E0 , E4 = E1 , E5 = E2
and drivers are 
X = TMC2209
Y = TMC5160 v1.2
Z0 = TMC5160 v1.2
Z1 = TMC5160 v1.2
Z2 = TMC5160 v1.2
Z3 = TMC5160 v1.2
E0 = TMC2209
E1 = TMC2209
E2 = TMC2209
Using 3 in 1 out Diamond hotend
I am attaching my configurations and pins files, Please check the problem
i am using Marlin version 2.0.5.3

[Configuration & Pins.zip](https://github.com/FLYmaker/FLYF407ZG/files/5299136/Configuration.Pins.zip)

FLYmaker, 2020-09-30:
I will upgrade you to version 2.6.0.1, please wait

FLYmaker, 2020-09-30:
You send me the compiled firmware, let me try

gsuresh2u, 2020-09-30:
[firmware.zip](https://github.com/FLYmaker/FLYF407ZG/files/5302974/firmware.zip)
Sir, attached my firmware
Please make sure drivers are explained in above post

FLYmaker, 2020-09-30:
I tested this firmware, the XY axis is normal

FLYmaker, 2020-09-30:
Maybe your driver is broken. A broken driver may cause damage to the transistor on the motherboard. Generally, the probability is not high.

FLYmaker, 2020-09-30:
What you need to do now is to make sure that a drive and motor are good, take out a new board, flash the firmware, test the X and Y axis, and don’t connect anything to other ports

gsuresh2u, 2020-09-30:
OK, I will try today

gsuresh2u, 2020-09-30:
Sir, Driver is Ok, its working fine, i checked with swapping with Z axis driver
so, its working 100% fine, i tried with my extra spare board also
Its also same problem
So i am thinking problem should be in firmware 
i will check everything and update you tomorrow
Thanks for support

gsuresh2u, 2020-10-01:
Finally i found problem in firmware
Original values are
 `#define DEFAULT_MAX_FEEDRATE          { 150, 150, 5, 25 }`
i changed to 
`#define DEFAULT_MAX_FEEDRATE          { 5, 5, 5, 25 }`
Then X and Y working fine, tried with old board also, then X is working fine Y has problem
tested with my spare board also, its also working fine

I have fixed bed at bottom of frame, it will not move 
when i try to Z home, its saying "Z Probe Past Bed"
my probe is working well, connected to Z MAX port on board
X and Y are homing fine
Please check the attached configuration, 
Upgraded to Marlin 2.0.6.1
[Configuration.zip](https://github.com/FLYmaker/FLYF407ZG/files/5309440/Configuration.zip)

FLYmaker, 2020-10-01:
You need to download version 2.0.6.1 of my github. But I will not set up auto-leveling, auto-leveling related issues, you need to study by yourself. very sorry

gsuresh2u, 2020-10-01:
Downloaded your version and changed as per my need
i am so much confused about Z Probe
i am not using Z Max and Z min endstops
Only using Z Probe as Z endstop, which is attached to Nozzle
To which port on Board need to connect from that Z Probe ?
Z Min or Z3 ?
attached my Configurations
[Configurations.zip](https://github.com/FLYmaker/FLYF407ZG/files/5312481/Configurations.zip)

FLYmaker, 2020-10-01:
Z3pin

gsuresh2u, 2020-10-04:
Solved my problem, there are 2 options 
i am posting my solutions , so it may be useful to someone

Option 1 : Connecting Z Probe to Z3 Pin
Disable `//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`
Disable `//#define USE_ZMIN_PLUG`
Enable `#define USE_PROBE_FOR_Z_HOMING`
then connect Z probe to Z3 Pin

Option 2 : Connecting Z Probe to Z Min Pin
Enable `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`
Enable `#define USE_ZMIN_PLUG`
Enable `#define USE_PROBE_FOR_Z_HOMING`
And you need to redefine Z min Pin
`#define Z_MIN_PROBE_PIN PF0`

I Have tried both options , its working fine

FLYmaker, 2020-10-04:
> 启用`#define USE_ZMIN_PLUG`

thank you