Mellow-3D/FLYF407ZG #29: BLTouch/3DTouch not responding

RemyBaligant, 2020-09-08:
Hi, 

I received a new board from Mellow, everything is working exept the servo out for BL/3DTouch.
The 3DTouch successfully respond on chitu3d controller.
I tried the last marlin release on your github.
Not possible to deploy or stow. The 3Dtouch do is self test at powering up, but after that, it seems it does not received pwm commands. 
I've not already plugged the oscilloscope to the servo signal pin, i'll try..

Any other bug reports for 3Dtouch users  ? 
Thanks. Best Regards

RemyBaligant, 2020-09-08:
I tried two different 3Dtouch, both working on chitu3d and mks robin nano v1.2. No success for now. I tried with 3.3Vdc and 5Vdc power supply for 3Dtouch.

FLYmaker, 2020-09-09:
The fan pwm conflicts with the servo timer. I have found the problem.

FLYmaker, 2020-09-09:
The fan pwm conflicts with the servo timer. I have found the problem. Need to disable here

FLYmaker, 2020-09-09:
You send your firmware source code here, I will send it to you after modification

RemyBaligant, 2020-09-09:
Thanks for support. Here's the zip containing the Marlin i use to test(last release):

https://drive.google.com/file/d/1nRm3v2nWciPvoIQC-2D-wXvdhR386qUE/view?usp=sharing

RemyBaligant, 2020-09-09:
i tested the servo output on a scope. I have 3.3V peak for PWM wich is fine. Probably a timing problem ?

FLYmaker, 2020-09-09:
[Marlin-2.0.x_FLYF407ZG.zip](https://github.com/FLYmaker/FLYF407ZG/files/5194086/Marlin-2.0.x_FLYF407ZG.zip)

FLYmaker, 2020-09-09:
[image]
I disabled PE13 here, which is the fan pin, and PE11 is the servo pin. Disable the fan PWM, the servo can be used normally. So the fan pwm and the servo pwm using the same timer will conflict

FLYmaker, 2020-09-09:
This is a bug

RemyBaligant, 2020-09-09:
Ok thanks for this fix i try it. 
So if i understand well, PE13 is FAN4. This one is totally disabled if servo is used ? 
Is it the only fan on the same timer ? 
I use fan0 for hotend cooling, fan1 for part cooling and fan5 for steppers cooling(activated when motion detected).

FLYmaker, 2020-09-09:
FAN4 can continue to be used, you can test it

RemyBaligant, 2020-09-09:
https://drive.google.com/file/d/1CP34IeChswxJHbPc4XPduP9-M-xJHSbl/view?usp=sharing

Thanks :) i can start printing test now

RemyBaligant, 2020-09-09:
Deploy and Stow work but when homing 3Dtouch deploy/stow a few times and then stop the printer... I try to add some delay. 300ms is maybe to short for 3dtouch..

RemyBaligant, 2020-09-09:
Ok, G28 is working great now with #define SERVO_DELAY { 750 }. 
Problem solved for 3DTouch ! :)