Mellow-3D/FLYF407ZG #30: This board supports 12V 50W heater ?

gsuresh2u, 2020-10-04:
Hi, i am using **12V 50W** heater for Hotend
when i tried to increase temperature upto 200, it going up to 175 only
then led near T0 mosfet is off, and temperature is showing constant 175 on screen, even if the temperature is down, its still showing the same 175, when i switch off and on again, then its showing correct temperature 
i calculated PID values with `M303 S150 C8` , 
then it gave me **Kp 147.07, Ki 17.18, Kd 214.74**
after updating firmware with the new values, then i tried again for PID values for 200 temperature 
M303 S200 C8,  then after sometime, its saying "Heating failed, Kill called"
i am trying this from 2 days, searched google, not found any solution, so i am asking here
Is mother board support **12V 50W** heater ?
there is no problem with heat bed, as i am using silicon heater of AC current, 2000W using solid state relay
here are my PID settings

```
#define PIDTEMP
#define BANG_MAX 230  //255   
#define PID_MAX BANG_MAX
#define PID_K1 0.7 //0.95

 #define DEFAULT_Kp 147.07
  #define DEFAULT_Ki 17.18
  #define DEFAULT_Kd 314.74
```
do i need to change heater to **40w** ?
Please help me, i am trying this from 2 days

gsuresh2u, 2020-10-05:
Sir, i have tried disabling //#define PIDTEMP
but still its not heating upto 200
only it is reaching upto 175, then led is off near mosfet
Please help me, i am stuck at this point from 3 days

FLYmaker, 2020-10-06:
Heating to 175 degrees after the motherboard has a warning error?

FLYmaker, 2020-10-06:
#define THERMAL_PROTECTION_HOTENDS    You can try to disable this

gsuresh2u, 2020-10-06:
After reaching 175, its not giving any error (i am using BTT TFT70 Screen)
simply led near mosfet of T0 is off after 175, temperature is showing same 175, even if its at room temperature until i restart board

Thanks for info` #define THERMAL_PROTECTION_HOTENDS`
i will disable and update you
Thank you

gsuresh2u, 2020-10-06:
Disabling `//#define THERMAL_PROTECTION_HOTENDS` works fine
now its heating excellent
Thank you so much

closed 2020-10-06 by gsuresh2u