Mellow-3D/FLYF407ZG #31: EXTRUDER_1 (Technically E4) Not working

gsuresh2u, 2020-10-06:
Sir, my Extruder 1 is not working, its connected to E4 port
E0 (connected to E3) and E2 (connected to E5) working well
there is no problem with Driver Or Motor (i checked with swapping with other driver and motor)
here is my configuration

`#define EXTRUDERS 3`
`#define SINGLENOZZLE` (Using Diamond Hotend)

```
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC5160
#define Z_DRIVER_TYPE  TMC5160
#define Z2_DRIVER_TYPE TMC5160
#define Z3_DRIVER_TYPE TMC5160
#define Z4_DRIVER_TYPE TMC5160
#define E0_DRIVER_TYPE TMC2209
#define E1_DRIVER_TYPE TMC2209
#define E2_DRIVER_TYPE TMC2209
```

My PINS Configuration
```
#define X_STEP_PIN                          PB9
#define X_DIR_PIN                           PE0
#define X_ENABLE_PIN                        PE1
#ifndef X_CS_PIN
  #define X_CS_PIN                          PG13
#endif

#define Y_STEP_PIN                          PB8
#define Y_DIR_PIN                           PG11
#define Y_ENABLE_PIN                        PG12
#ifndef Y_CS_PIN
  #define Y_CS_PIN                          PG10
#endif

#define Z_STEP_PIN                          PA8
#define Z_DIR_PIN                           PD6
#define Z_ENABLE_PIN                        PD7
#ifndef Z_CS_PIN
  #define Z_CS_PIN                          PD5
#endif

#define Z2_STEP_PIN                         PC7
#define Z2_DIR_PIN                          PD3
#define Z2_ENABLE_PIN                       PD4
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN                         PD1
#endif

#define Z3_STEP_PIN                         PC6
#define Z3_DIR_PIN                          PA15
#define Z3_ENABLE_PIN                       PD0
#ifndef Z3_CS_PIN
  #define Z3_CS_PIN                         PA14
#endif

#define Z4_STEP_PIN                         PD15
#define Z4_DIR_PIN                          PG7
#define Z4_ENABLE_PIN                       PG8
#ifndef Z4_CS_PIN
  #define Z4_CS_PIN                         PG6
#endif

#define E0_STEP_PIN                         PD14
#define E0_DIR_PIN                          PG4
#define E0_ENABLE_PIN                       PG5
#ifndef E0_CS_PIN
  #define E0_CS_PIN                         PG3
#endif

#define E1_STEP_PIN                         PD13
#define E1_DIR_PIN                          PD11
#define E1_ENABLE_PIN                       PG2
#ifndef E1_CS_PIN
  #define E1_CS_PIN                         PD10
#endif

#define E2_STEP_PIN                         PD12
#define E2_DIR_PIN                          PD8
#define E2_ENABLE_PIN                       PD9
#ifndef E2_CS_PIN
  #define E2_CS_PIN                         PB12
#endif
```

i have not understand its problem
In display (BTT TFT70) also, its not showing E1 (steps per mm menu)
Is the problem in Firmware ? attached my config files and pins file
[Configurations.zip](https://github.com/FLYmaker/FLYF407ZG/files/5333734/Configurations.zip)

and here is the video link
http://www.mediafire.com/file/1mo7spj8bob3ggf/EXTRUDER_ERROR.mp4/file

FLYmaker, 2020-10-06:
You need to send the mainboard T1 to switch to the extruder 1, send T2 to switch to the extruder 2

FLYmaker, 2020-10-06:
![Uploading 4F}$F)9HILFFT~3$A%QO][4.png…]()

FLYmaker, 2020-10-06:
[image]

gsuresh2u, 2020-10-06:
Sir, i tried swapping E1 and E2 pins like below,
now E0, E1 are working fine, E2 is not working
I think its board problem 
```
#define E1_STEP_PIN               PD12 
#define E1_DIR_PIN                 PD8 
#define E1_ENABLE_PIN          PD9 
#ifndef E1_CS_PIN
  #define E1_CS_PIN                PB12
#endif

#define E2_STEP_PIN               PD13
#define E2_DIR_PIN                 PD11
#define E2_ENABLE_PIN          PD2
#ifndef E2_CS_PIN
  #define E2_CS_PIN                 PB10
#endif
```

FLYmaker, 2020-10-06:
Please send your complete firmware. I will test it tomorrow

gsuresh2u, 2020-10-07:
Sir, i have attached full firmware folder, i have deleted pio folder files to reduce size
Today other problem came up, problem is Bed temperature is showing 45 at room temperature, 
hotend temperature is showing 28, which is same as room temperature
i disconnected bed temperature sensor from board, not is showing 28
swapped bed sensor and hotend sensor, now hotend sensor is notnal, bed temperature is showing same as 45
What is happening this board ?

[Uploading English-firmware.zip…]()

gsuresh2u, 2020-10-07:
Sir, i have attached full firmware folder, i have deleted pio folder files to reduce size
Today other problem came up, problem is Bed temperature is showing 45 at room temperature,
hotend temperature is showing 28, which is same as room temperature
i disconnected bed temperature sensor from board, now it is showing 28
swapped bed sensor and hotend sensor, now hotend sensor is normal, bed temperature is showing same as 45
What is happening this board ?
[English-firmware.zip](https://github.com/FLYmaker/FLYF407ZG/files/5337758/English-firmware.zip)

gsuresh2u, 2020-10-07:
Sir, today i have tested the same firmware with my spare board
the same E1 is not working, so it should be firmware problem
i think there is no chance that 2 boards had same complaint

FLYmaker, 2020-10-07:
Hello, I have carried out a test and all three extruders can work normally

FLYmaker, 2020-10-07:
[Uploading firmware.zip…]()

FLYmaker, 2020-10-07:
[firmware.zip](https://github.com/FLYmaker/FLYF407ZG/files/5340620/firmware.zip)

FLYmaker, 2020-10-07:
You first drive the following jumper cap, inserted into the normal mode for testing

gsuresh2u, 2020-10-07:
Can i use firmware.bin you uploaded , has the same drivers i used ?

FLYmaker, 2020-10-07:
I'm using TMC2208, normal mode

gsuresh2u, 2020-10-07:
Uploaded your firmware and changed E0, E1, E2 pins to Normal mode
then all3 working fine, have you changed any settings in firmware ?
and one of my board Bed temperature sensor is not working, showing some value even any sensor is not connected to it
how to assain bed temperature pin to any other free temperature pins ?

gsuresh2u, 2020-10-07:
Please upload the config files you compiled

FLYmaker, 2020-10-07:
[Uploading English-firmware.zip…]()

FLYmaker, 2020-10-07:
[English-firmware.zip](https://github.com/FLYmaker/FLYF407ZG/files/5340956/English-firmware.zip)