Mellow-3D/FLYF407ZG #32: Board does a reset while homing

gsart84, 2020-12-22:
Hello

Got here new board from Mellow

After reading all information here, now i have a question.

i am using many motors X1, X2,Y1, Y2, Z1, Z2, E0, E1 for my special construction

Wheni move z: serial connection gets lost, when i do g28, first X1, X2, and Y work perfect. then Z movement causes reboot of Board

i also use sensorless homing on X and Y.

Have mounted BL-Touch

[Marlin.zip](https://github.com/FLYmaker/FLYF407ZG/files/5730429/Marlin.zip)

gsart84, 2020-12-23:
i beleve i found the Problem.

i use dual Z motors with a M5 Leadscrew (pitch 0,8) what means with 256 microsteps it results in 64000 steps. this let
the board crash and reboot after 10mm movement in z.
calculated all down and with 32000 steps and 128 microsteps in configuration. it seems to work.

with previously Rumba32 Board the 64000 steps was no problem. i beleve FLY TMC2209 are not same perfect quality.

FLYmaker, 2020-12-25:
Sorry to work too much lately, I forgot to reply, I will test thanks for the feedback